Wicker25/Rpi-hw #10: source build problem

Paku-, 2015-01-15:
Hi,

Trying build the master using:
    
    cmake .. -DCMAKE_INSTALL_PREFIX=/usr -DUSE_FREETYPE=OFF -DUSE_MAGICK=OFF
    makeinstall

but I ma stacked on:
    
    cc1plus: error: unrecognized command line option ‘-std=c++11’
    

any ideas ??

Sure did not:
    
    apt-get install g++-4.7 cmake libfreetype6-dev libgraphicsmagick++1-dev doxygen
    

but have the compiler and cmake and disabled the rest (I hope)

Wicker25, 2015-01-15:
Hi @Paku-,

what version of g++ you are using? Raspbian's default compiler is g++ 4.6, but you need to install the 4.7 version.

Regards,
Giacomo

Paku-, 2015-01-15:
thanks will check this out

Paku-, 2015-01-15:
that's

    gcc version 4.6.3 (Debian 4.6.3-14+rpi1)

sorry ...

Wicker25, 2015-01-15:
No problem, @Paku- ;)

Regards,
Giacomo

closed 2015-01-15 by Wicker25

Paku-, 2015-01-15:
One more question ...

Should I just add gcc-4.7 or first remove gcc package ??

after gcc removal got little mess ... gcc-4.7 install makes me compiler less :)

Wicker25, 2015-01-15:
You can have both versions installed without problem. Install the new version of g++ by using `apt-get` and set the 4.7 as default version (see https://github.com/Wicker25/Rpi-hw/wiki/Download%20and%20install)

Paku-, 2015-01-15:
Got it thanks ...