Wicker25/Rpi-hw #11: Bashfile Installer?

Annonomus-Penguin, 2015-05-30:
I'm curious why you don't have some sort of bashfile installer for this to make it easier for newer users. Would you accept a PR for this?

For example, this could be done:

```bash
#!/bin/bash
sudo apt-get install g++-4.7 cmake libfreetype6-dev libgraphicsmagick++1-dev -y
wget https://github.com/Wicker25/Rpi-hw/archive/master.tar.gz -O rpi-hw.tar.gz
tar -zxvf rpi-hw.tar.gz
cd Rpi-hw*
cmake . -DCMAKE_INSTALL_PREFIX=/usr
make
sudo make install
cd ../
sudo rm -rf Rpi-hw
rm rpi-hw.tar.gz
```

Haven't tested it yet and I probably missed something there, but it took me literally 2 minutes to copy and paste that from the **[install](https://github.com/Wicker25/Rpi-hw/wiki/Download%20and%20install)** page.

Wicker25, 2015-06-02:
Hi @Annonomus-Penguin,

Wicker25, 2015-06-02:
Hi @Annonomus-Penguin,

when I wrote the library I thought I would have enough time to build some installer packages (deb, rpm, etc) but I was wrong.

Your script is only for Debian-based distros, but if you want to share it we can add it to the documentation. What do you think about it?

Annonomus-Penguin, 2015-06-03:
@Wicker25 I just pulled it **[from your documentation](https://github.com/Wicker25/Rpi-hw/wiki/Download%20and%20install)** :p