Wicker25/Rpi-hw #14: raspberry Pi 2 Compability

tolgamyth, 2015-08-02:
compiled rpi-hw code on raspberry B+ works for the keypad example but not on raspberry pi 2. We connected all the pins same as on B+ but not worked on pi2.
 Any ideas?

Wicker25, 2015-08-02:
Hi @tolgamyth ,

I'm sorry but I developed Rpi-hw for Raspberry Pi version 1 and actually I haven't enough time to update it to support Raspberry 2.

Regards,
Giacomo

tolgamyth, 2015-08-02:
i suspect that you use driver bcm2835 but pi2 has 2836(at least i my pi has that version) any way to find that drive or could you add it in the code?

tolgamyth, 2015-08-02:
i replaced below line
#define BCM2708_PERI_BASE	0x20000000 in bcm2835.hpp to
#define BCM2708_PERI_BASE	0x3F000000 and it worked. 
B+  512Mb but pi2 is 1Gb that was the reason. 
thanks however maybe you can add an option to select between raspberry versions on your future releases.

closed 2015-08-02 by tolgamyth