Wicker25/Rpi-hw #15: Raspberry pi 2

kohoutovice, 2015-10-11:
Hello,
I really love your library. I using it with lcd pcd8544 display and it great, but now i start using raspberry pi 2 and on this new raspberry it doesn't work so i tried to change (in this file https://github.com/Wicker25/Rpi-hw/blob/master/include/rpi-hw/driver/bcm2835.hpp ) bcm peri base to this #define BCM2708_PERI_BASE	0x3F000000 and it work but it work bad, there are lots of bug in communication (i think) because the picture isn't consistent. So don't you now how to run it properly on the raspberry pi 2.
Thank you for your answers.

Wicker25, 2015-10-11:
Hi @kohoutovice,

I'm very sorry for this, but I don't have a Raspberry Pi 2 yet and I cannot update my library.

Does the examples provided with Rpi-hw work properly? What version of Rpi 2 are you using?

kohoutovice, 2015-10-16:
The examples (from this site http://hackyourmind.org/articles/2013/06/22/raspberry-pi-with-philips-pcd8544.html) sometimes work properly and sometimes not (maybe 50% are bad). When i use this example ( for example the first exaple with drawing some lines and circles) i get this outputs (it's the same program)

[image]
[image]
[image]
[image]

I have the raspberry pi 2 model B.

Wicker25, 2015-10-17:
@kohoutovice , maybe the working frequency is too high for your display. Can you try to increase the delay value at the end of the following line in my library?

https://github.com/Wicker25/Rpi-hw/blob/master/src/display/pcd8544.cpp#L35

Try to put 20000 instead of 10000. Once you do this you can recompile the library and make a new attempt ;)