Wicker25/Rpi-hw #19: Compile error

girst, 2017-01-25:
I tried to build the library with 
```
cmake . -DCMAKE_INSTALL_PREFIX=/usr  -DUSE_FREETYPE=OFF -DUSE_MAGICK=OFF
make
```
but it won't compile. 

```
Scanning dependencies of target rpihw
[  3%] Building CXX object CMakeFiles/rpihw.dir/src/utils.cpp.o
[  6%] Building CXX object CMakeFiles/rpihw.dir/src/time.cpp.o
[  9%] Building CXX object CMakeFiles/rpihw.dir/src/gpio.cpp.o
[ 12%] Building CXX object CMakeFiles/rpihw.dir/src/driver/bcm2835.cpp.o
[ 15%] Building CXX object CMakeFiles/rpihw.dir/src/driver/spi.cpp.o
[ 18%] Building CXX object CMakeFiles/rpihw.dir/src/driver/i2c.cpp.o
In file included from /home/pi/libs/Rpi-hw/include/rpi-hw/driver/i2c.hpp:29:0,
                 from /home/pi/libs/Rpi-hw/src/driver/i2c.cpp:25:
/usr/include/linux/i2c-dev.h:37:8: error: redefinition of ‘struct i2c_msg’
 struct i2c_msg {
        ^
In file included from /home/pi/libs/Rpi-hw/include/rpi-hw/driver/i2c.hpp:28:0,
                 from /home/pi/libs/Rpi-hw/src/driver/i2c.cpp:25:
/usr/include/linux/i2c.h:68:8: error: previous definition of ‘struct i2c_msg’
 struct i2c_msg {
        ^
In file included from /home/pi/libs/Rpi-hw/include/rpi-hw/driver/i2c.hpp:29:0,
                 from /home/pi/libs/Rpi-hw/src/driver/i2c.cpp:25:
/usr/include/linux/i2c-dev.h:89:7: error: redefinition of ‘union i2c_smbus_data’
 union i2c_smbus_data {
       ^
In file included from /home/pi/libs/Rpi-hw/include/rpi-hw/driver/i2c.hpp:28:0,
                 from /home/pi/libs/Rpi-hw/src/driver/i2c.cpp:25:
/usr/include/linux/i2c.h:128:7: error: previous definition of ‘union i2c_smbus_data’
 union i2c_smbus_data {
       ^
CMakeFiles/rpihw.dir/build.make:182: recipe for target 'CMakeFiles/rpihw.dir/src/driver/i2c.cpp.o' failed
make[2]: *** [CMakeFiles/rpihw.dir/src/driver/i2c.cpp.o] Error 1
CMakeFiles/Makefile2:99: recipe for target 'CMakeFiles/rpihw.dir/all' failed
make[1]: *** [CMakeFiles/rpihw.dir/all] Error 2
Makefile:127: recipe for target 'all' failed
make: *** [all] Error 2
```

Raspberry Pi image from 2017-01-11, using g++4.7, last commit db4d9c0dd9d765a3f92f63b6aa413316f01be15e

girst, 2017-01-25:
removing line 28 in `include/rpi-hw/driver/i2c.hpp` fixes the compile error.

i have not checked if it breaks something else though.

girst, 2017-01-27:
I found an explaination [here](https://github.com/ceolin/soletta/blob/e8303d68b54b7925bef61f6f93d6b2d6bb6e7168/src/lib/io/sol-i2c-linux.c#L38-L48).

Wicker25, 2017-01-27:
Hi @girst , thank you very much! Unfortunately my library has not been very successful, so I'm not dedicating much time to this project. Anyway I will merge your PR as soon as possible!